marqs85/ossc #112: OSD info button doesn't work when disabled

filevans, 2024-04-25:
Hello, 
So as not to have the OSD info coming up on the screen during games that switch resolution, eg. Dead or alive 2, then the only way seems to be settings OSD status display to off? But when doing that it also stops you from being able to manually check the screen status / resolution using the info button. You should still be able to manually check the display info / resolution by pressing the info button on the remote but you can't

marqs85, 2024-11-01:
Ossc Pro v0.77 just added a menu option to display a sticky info window, but code space on original OSSC is fully utilized so this won't make it there.

closed 2024-11-01 by marqs85

filevans, 2024-11-02:
Seems ridiculous that you can't manually display resolution info on the screen even with automatic osd off, you don't want it keep displaying frequent resolution changes on games such as dead or alive. Is there any work around?

How about changing it so you can have an OSD setting of 0 seconds instead of off? Then you could manually press it on when you want? That would resolve the problem?

OSD 0 seconds... That would fix the problem

filevans, 2024-12-06:
no, I said you can add a setting that is: OSD info **0 seconds**, rather than "off", then it won't automatically display OSD upon resolution changes unless you manually display it, but doesn't have to be fully turned off so resolution info OSD is still manually accessible