marqs85/ossc #113: Autodetect input get stuck on AV3 (firmware 1.09)

FluffyRedLobster, 2024-06-04:
When auto-scanning all inputs after the test pattern, if no input is connected the scan gets as far as AV3 RGsB, the LCD flashes up an input frequency then the auto-scan gets “stuck” on the AV3 inputs – it doesn’t loop back to AV1.  Each time it loops round to AV3 RGsB it flashes up a frequency momentarily, usually either 11kHz or 21kHz.  

Steps to repro:
1. Power off any active inputs
2. Install firmware 1.09
3. Set "Autodetect input" to "All inputs" and Save config
4. Reboot, wait for no-input image
5. Press button 0 to start input seek

Expected result:
1. AV1 - AV3 should scan for an active input, returning to AV1 and looping round if no active input is found.

Actual result:
1. When scan reaches AV3 RGsB, the LCD flashes up an input frequency then the auto-scan gets “stuck” on the AV3 inputs.  Video: https://www.icloud.com/photos/#/icloudlinks/062olIDHPf5zEEsFDn_sG9KKQ/0/

Notes - this was reproduced on a Kaico 1.6 board with hardware mod performed.  It has also been reported on a 1.8 board (source of this board not reported).  It is not reproducible on firmware 1.08 or earlier.

See also: https://videogameperfection.com/forums/topic/ossc-v1-xx-series-beta-firmware/page/16/#post-61817

Pu5hky, 2024-06-05:
I'm Pushky from videogameperfection boards, and yes i can get behind this issue, my ossc v1.6 modded as this problem, and i thought it was the mod, but after checking everything was ok, i bought a v1.8 to test out, it came with v1.07 firmware, worked nicely, but after the update to v1.09 the issue started happening on that ossc aswell. 
I thought i was alone or it would be some problem with some device i had in the chain of devices. But i think it's safe to assume that the sync changes in the v1.09 firmware could have something to do with it.

marqs85, 2024-06-05:
The issue is caused by change of physical sync input which can trigger a pulse on VSYNC line that is subsequently detected as sync activity. The activity resets timeout for autodetection so it never stops. All v1.xx firmwares are affected, but on v1.09 this happens more frequently so instead of looping endlessly thru AV1-AV3, the autodetection is consistently reset on AV3 input so it restarts the loops there. A fix for this is under testing.

Pu5hky, 2024-06-05:
That's awesome. Thank you so much!

FluffyRedLobster, 2024-06-05:
Yes, thanks from me also!  If you need a beta tester prior to public release, happy to help.

marqs85, 2024-06-06:
You can download latest dev firmware with the fix from [here](http://www.infocult.com/m/tmp/ossc_1.10-luma.bin).

FluffyRedLobster, 2024-06-07:
Thanks @marqs85 - I've given it a quick test and from a first pass it resolves the issue for me.  I will play with it more over the weekend and connect as many different sources as I can to ensure all is still good generally.

Pu5hky, 2024-06-07:
Well, was on my lunch break and quickly tried the firmware. On the first run of the tests, it's working great. With the auto-detect enabled to scroll through all inputs it as no issues. My chain is as such: MD and TG16 on av1 through a switch Wii and ps2 on av2 through a switch and Dreamcast on av3. Quickly alternating between all of them goes smooth. Once again thank you

FluffyRedLobster, 2024-06-11:
Sorry for the slow update.  I've hooked up various sources through different inputs (PAL Gamecube through component, PAL Dreamcast through VGA, PAL Megadrive, PSX and N64 through Scart) - no issues with any of them that I can see.

Only thing I haven't tried is my old DOS PC / VGA mode 13 etc.  I can try that if needed though.

Looks good to me!

marqs85, 2024-06-11:
Fix included in 10f1afd79c808e4a297b6a464638b8d650653768, closing the issue.

closed 2024-06-11 by marqs85