marqs85/ossc #115: Official fw1.10 C64 Lumacode palette

TheRetroChannel, 2024-06-24:
Hey Markus, I just tested out the official 1.10 firmware. Great to see Lumacode is now officially supported for certain systems. But I noticed the colour palette for the C64 was reverted back to the one used in the first Lumacode test release.

Any chance this can be changed with the next release to use the palette from the final Lumacode test fw?

Cheers,

Mark

marqs85, 2024-06-25:
Seems like this got accidentally reverted during integration of various changes. Will be fixed on next firmware which will be out relatively soon as NES Lumacode support could be now added too.

marqs85, 2024-06-29:
Fixed in 82c498ac020229877ead6ef9bce6ac1010ad9dbd

closed 2024-06-29 by marqs85