marqs85/ossc #118: Ossc and Sony Bravia

pkerga, 2024-09-26:
Hello
I have an ossc 1.6 and a Sony Bravia TV.

I can get a lot of things to work in X2. But nothing with my AES, MVS or Neogeo CD. While it works on another TV. I read that the Bravias were quite capricious when it came to HDMI inputs. I have a VGA input on my TV but with an HDMI -> VGA adapter, the image is shaky.

Is there a setting I forgot? Does the mod to ossc 1.8 with firmware 1.xx change anything?
Could something like Dr HDMI could help ?
https://hdfury.com/product/dr-hdmi/
Thanks !!!

marqs85, 2024-09-30:
Neo Geos (especially MVS) have lower refresh rate than standard 59.94/60Hz sources. They also output 264 lines instead of typical 262/263. There's very little OSSC (regardless of HW revision) can do if a TV doesn't accept the modes, but you could check if shakiness with HDMI->VGA conversion is due to the adapter or TV's incompatibility by testing on another VGA screen.

pkerga, 2024-10-13:
I'm thinking of buying an ossc pro. Should this kind of issue still be there with the pro ?

marqs85, 2024-10-14:
Pro is able to normalize line count to CEA standard, and also refresh rate to 59.94/60Hz if needed.