marqs85/ossc #120: Always "No Sync" on YPbPr 480p AV2 input post-0.90

InfiniteBlueGX, 2024-11-17:
The YPbPr AV2 input seems to be broken for 480p sources in all builds post v0.90. I've tested this with a confirmed working NTSC Wii with component cables.

v1.11 always gives "No Sync" when connected via the YPbPr ports with a working signal. All other 1.X firmwares seem to have the problem as well. All were left at default settings.

Downgrading to v0.90 resolves the problem immediately, the picture is correctly displayed (default settings).

Apologies if this is already known - if you need me to do any additional testing, just let me know and I'm happy to help.

InfiniteBlueGX, 2024-11-29:
It's a v1.6 board, bought from VGPerfection in 2018.

I tried a PS2 connected via AV1 RGBs in 480i - same problem. Picture displays normally in v0.90 firmware at default settings, upgrading to v1.11 at default settings shows "NO SYNC" despite the console still being powered on and connected. Switching between passthru and Line2x modes has no effect either in v1.11.

marqs85, 2024-11-30:
Unmodified v1.6 board does not work with v1.xx firmware, ref. [this](https://videogameperfection.com/forums/topic/ossc-v1-xx-series-beta-firmware/).

closed 2024-11-30 by marqs85